jgauchia/IceNav-v3 #87: Disable Compass or define map orientation priority

hpsaturn, 2023-09-02:
## Summary

Sometimes the compass is not calibrated or it fails for magnetic fields like Tram lines. For this reason the orientation sometimes it's wrong or has a lot of noise. It's possible give more priority to the navigation calculation, or have the setting to disable the compass in the map?

jgauchia, 2023-09-03:
Disable the compass on the map?  or disable map rotation keeping the compass indicator located in the top right corner of the map?

hpsaturn, 2023-09-03:
Yes, I agree, disable the rotation of the map by compass, and the possibility to enable the rotation by heading or movement detection? (and yes, the compass in the corner will be nice to have)

jgauchia, 2023-09-03:
Ok, I'll try to do it! thanks @hpsaturn

jgauchia, 2023-09-03:
Issue in 0.1.8_dev branch

closed 2023-09-03 by jgauchia