jgauchia/IceNav-v3 #99: [Makerfab] checkbox settings freezing

hpsaturn, 2023-11-03:
## Overview

On the makerfabs board, some settings widgets are freezing the device. 

## Details

Raplication: settings->map settings->show compass

## Log output

```cpp
[   501][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   522][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=38 scl=39 freq=100000
[   523][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  4534][V][Preferences.cpp:352] getUChar(): nvs_get_u8 fail: Map_compass NOT_FOUND
[  4535][V][Preferences.cpp:352] getUChar(): nvs_get_u8 fail: Map_speed NOT_FOUND
[  4538][V][Preferences.cpp:352] getUChar(): nvs_get_u8 fail: Map_scale NOT_FOUND
[  4546][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Coords_X NOT_FOUND
[  4552][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Coords_Y NOT_FOUND
[  4560][V][Preferences.cpp:352] getUChar(): nvs_get_u8 fail: Map_vector NOT_FOUND
[  4567][V][preferences.h:91] load_preferences(): COMPASS OFFSET X  27.231785
[  4573][V][preferences.h:92] load_preferences(): COMPASS OFFSET Y  24.160965
[  4580][V][preferences.h:93] load_preferences(): MAP ROTATION 1
[  4586][V][preferences.h:94] load_preferences(): DEFAULT ZOOM LEVEL 17
[  4592][V][preferences.h:95] load_preferences(): SHOW MAP COMPASS 0
[  4598][V][preferences.h:96] load_preferences(): SHOW MAP SPEED 0
[  4604][V][preferences.h:97] load_preferences(): SHOW MAP SCALE 0
[  4610][V][preferences.h:98] load_preferences(): GPS SPEED 2
[  4616][V][preferences.h:99] load_preferences(): GPS UPDATE RATE 3
[  4622][V][preferences.h:100] load_preferences(): COMPASS POS X 52
[  4628][V][preferences.h:101] load_preferences(): COMPASS POS Y 91
[  4634][V][preferences.h:102] load_preferences(): COORDINATE POS X 15
[  4640][V][preferences.h:103] load_preferences(): COORDINATE POS Y 10
[  4646][V][preferences.h:104] load_preferences(): SPEED POS X 11
[  4652][V][preferences.h:105] load_preferences(): SPEED POS Y 339
[  4658][V][preferences.h:106] load_preferences(): VECTOR MAP 0
[  4671][V][sdcard.h:35] init_sd(): SD Card Mounted
[  4771][V][sdcard.h:49] init_SPIFFS(): SPIFFS Mounted
[  5347][E][Wire.cpp:137] setPins(): bus already initialized. change pins only when not.
[  5347][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  5351][V][Touch_FT5x06.cpp:63] _check_init(): [FT5x06] CIPHER:0x64 / FIRMID:0x03 / VENDID:0x11
E (5088) ledc: ledc_get_duty(740): LEDC is not initialized
[ 21784][V][button_bar.h:26] settings(): Settings

```

jgauchia, 2023-11-03:
Freezes in settings screen or in map screen?

hpsaturn, 2023-11-03:
In map settings when you try to check the checkbox widget of "show compass".

I reached to capture a better log trace, but I think it is not enough. I attached it here:
[output.txt](https://github.com/jgauchia/IceNav-v3/files/13255213/output.txt)

jgauchia, 2023-11-03:
Please , check modifications in branch associated to this issue

hpsaturn, 2023-11-05:
Thanks, with the last commit it was fixed.

closed 2023-11-05 by hpsaturn