jgauchia/IceNav-v3 #102: how can i compile it on by vscode with platformio of esp32s3 borad.

zhjygit, 2024-01-06:
Hi sir, my board is esp32s3, if i compile this code with original CUSTOMBOARD, i cannot upload to my device with the error "it is a esp32s3 not a esp32"
If i make change the platformio.ini as follows:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
description = ESP32 GPS Navigator
default_envs = esp32-s3-devkitm-1

[common]
platform = espressif32
framework = arduino
version = 0.1.7
revision = 38
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_echo = yes
monitor_filters = 
  send_on_enter
  esp32_exception_decoder
extra_scripts = pre:prebuild.py
build_flags = 
	-D CORE_DEBUG_LEVEL=5
	-D USE_LINE_BUFFER=1
	-D DISABLE_RADIO=1
	-D GPS_BAUDRATE=9600
	-D AT6558D_GPS=1
	-D MULTI_GNSS=1
	-D BAUDRATE=115200
	-D DEBUG=1
lib_deps = 
	mikalhart/TinyGPSPlus@^1.0.3
	paulstoffregen/Time@^1.6.1
	; me-no-dev/ESP Async WebServer@^1.2.3
	lvgl/lvgl@^8.3.9
	lovyan03/LovyanGFX@^1.1.8
	jchristensen/Timezone@^1.2.4

[esp32_common]
platform = ${common.platform}
framework = ${common.framework}
monitor_speed = ${common.monitor_speed}
monitor_rts = ${common.monitor_rts}
monitor_dtr = ${common.monitor_dtr}
monitor_echo = ${common.monitor_echo}
monitor_filters = ${common.monitor_filters}
build_flags = ${common.build_flags}
lib_deps = ${common.lib_deps}
extra_scripts = ${common.extra_scripts}

[env:esp32-s3-devkitm-1]
extends = esp32_common
board = esp32-s3-devkitm-1
upload_port = /dev/ttyUSB0
board_upload.flash_size = 16MB
board_build.partitions = default_16MB.csv
lib_deps = 
	${common.lib_deps}
	adafruit/Adafruit Unified Sensor@^1.1.13
	adafruit/Adafruit HMC5883 Unified@^1.2.1
	adafruit/Adafruit BusIO@^1.14.3
	adafruit/Adafruit BME280 Library@^2.2.2
	; me-no-dev/ESP Async WebServer@^1.2.3 ; Why we need that? This library is heavy (~200kb)
build_flags = 
	${common.build_flags}
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-D ENABLE_COMPASS=1
	-D ENABLE_BME=1
	-D TFT_WIDTH=320
	-D TFT_HEIGHT=480
	-D TFT_BL=33

[env:MAKERF_ESP32S3]
extends = esp32_common
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
lib_deps =
  ${common.lib_deps}
  bolderflight/Bolder Flight Systems MPU9250@^1.0.2
build_flags = 
	${common.build_flags}
	-DBOARD_HAS_PSRAM
	-D ENABLE_COMPASS=1
	; -D ENABLE_GYRO=1
	; -D ENABLE_BME=1
	-D TFT_WIDTH=320
	-D TFT_HEIGHT=480
	-D TFT_BL=45

[image]

How can i change the ini file for my esp32se-groom-1 with 4.3 lcd

jgauchia, 2024-01-06:
Hi, 
The .ini file seems to be ok.
Maybe is a problem with Python.
Later I'll try to compile It with your .ini

jgauchia, 2024-01-08:
Hi
Sorry, I'm on holydays. This week I'll test it and give you a answer

jgauchia, 2024-01-20:
I've created a new branch to test your configuration. Today, I will run the tests to see if it compiles.

jgauchia, 2024-01-20:
> > Hi, The .ini file seems to be ok. Maybe is a problem with Python. Later I'll try to compile It with your .ini
> 
> Hi sir, how about the test on the esp32s3 board,waiting for you reply.

Hi, 
What is the pin configuration for your TFT and SD so that I can include it and test if it compiles correctly?

jgauchia, 2024-01-20:
Please test this branch:

[https://github.com/jgauchia/IceNav-v3/tree/esp32s3-board](https://github.com/jgauchia/IceNav-v3/tree/esp32s3-board)

SD Card and TFT have the same GPIO's than CUSTOMBOARD.

This branch seems to compile ok.

jgauchia, 2024-01-20:
> > Hi, The .ini file seems to be ok. Maybe is a problem with Python. Later I'll try to compile It with your .ini
> 
> Hi sir, how about the test on the esp32s3 board,waiting for you reply.

Please test this branch:

[https://github.com/jgauchia/IceNav-v3/tree/esp32s3-board](https://github.com/jgauchia/IceNav-v3/tree/esp32s3-board)

SD Card and TFT have the same GPIO's than CUSTOMBOARD.

This branch seems to compile ok.

zhjygit, 2024-01-22:
> > > Hi, The .ini file seems to be ok. Maybe is a problem with Python. Later I'll try to compile It with your .ini
> > 
> > 
> > Hi sir, how about the test on the esp32s3 board,waiting for you reply.
> 
> Hi, What is the pin configuration for your TFT and SD so that I can include it and test if it compiles correctly?

Hi sir , here is my board sheet:
[4.3寸开发板原理图.pdf](https://github.com/jgauchia/IceNav-v3/files/14004025/4.3.pdf)

zhjygit, 2024-01-22:
[image]
It is esp32s3 4.3 GT911 touch lcd .

closed 2024-04-18 by jgauchia