jgauchia/IceNav-v3 #104: what iot device shoud i connect?

zhjygit, 2024-04-12:
Hi sir,this project is great, i connect a gps moudule with gpio 17 and gpio 18, i put maps with (13 14 15 16 resolutions) into sd card.
I can get the gps location and display maps on the screen, i can see the speed of my foot about 3-6 km/h.
How the speed data display?i don't connect anything like move sensors to the esp32s3 borad but gps modules.
Of course,the speed display is not accurate， and the maps zoom in or zoom out is slow between 13 14 15 and 16 resolution maps.
Is necessary to buy the HCM5883 + MPU6050+MPU9250, what the difference between them?
Waiting for you reply.

jgauchia, 2024-04-12:
> Hi sir,this project is great, i connect a gps moudule with gpio 17 and gpio 18, i put maps with (13 14 15 16 resolutions) into sd card.
> I can get the gps location and display maps on the screen, i [...]

Hi, 
The speed data IS from GPS data, no extra module is added.
The HCM5883 Or MPU are compass module, isn't necessary, but if you want to have an Compass in the program you need one of them for example HCM5883
The speed of the Maps maybe is problem that you have enabled Compass module and don't have any. Please comment the line -D ENABLE_COMPASS in esp32s3 section in platformio.ini and try again

zhjygit, 2024-04-14:
Thanks sir. I have bought a mpu9255(instead of mpu9250).
I doubt that, the speed of owner of esp32s3, that is to say, when i run or walk with device esp32s3, the speed on the screen is like 5 km/h, and the speed is infact from the mpu9255 module with faster and preciser record; meanwhile, the compass arrows on the maps is from GPS module and update with the move of GPS module, is is right?
In summary,  the modue mpu9255 is for human speed record, the gps module is for location on the map.
The problem is that, how could i get the memory occupied and optimize the memory?

jgauchia, 2024-04-18:
Hi, 
No, now speed is from GPS module, not from any IMU or similar, and compass arrow was from IMU or magnetometer module. If you want speed from IMU or similar, feel free to make a PR from my code and code it.

That is:
* GPS -> Current position and speed
* IMU or magnetometer -> current degrees like compass 

> "The problem is that, how could i get the memory occupied and optimize the memory"  

Sorry, I don't understand this question, what's the problem?

zhjygit, 2024-04-19:
Thanks,sir, i will try.

zhjygit, 2024-04-19:
The memory problem is that, when the device boots , lcd shows that "mem 300k, free, 90%", i only have 300k memory to use, is that right?

jgauchia, 2024-04-19:
This 300k are RAM , ESP32S3 (if you have makerfabs board) has an extra 2Mb PSRAM sufficient to handle the program.

Map images, and all the GUI interface uses PSRAM . 

My board has 180k of free RAM , and uses aprox 1,5Mb of PSRAM.

Why do you need extra memory?

zhjygit, 2024-04-22:
I want to finish location share.There will be mang other esp32s3 gps arrow display on the map.
maybe esp-now can share locations between devices.
so I worry about the memory limit.
could you give me some advice?

jgauchia, 2024-05-15:
Sharing location isn't planned to be implemented for now. At this point, I can't help you with that.

closed 2024-05-15 by jgauchia