jgauchia/IceNav-v3 #105: IMU initialization unsuccessful

zhjygit, 2024-04-18:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   504][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   509][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=38 scl=39 freq=100000
[   510][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   706][E][compass.h:65] init_compass(): IMU initialization unsuccessful
[   706]��[   714][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: C_offset_x NOT_FOUND
[   714][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: C_offset_y NOT_FOUND
[   719][V][preferences.h:22] load_preferences(): OFFSET X  0.000000
[   726][V][preferences.h:23] load_preferences(): OFFSET Y  0.000000
[   739][V][sdcard.h:35] init_sd(): SD Card Mounted
[   839][V][sdcard.h:49] init_SPIFFS(): SPIFFS Mounted
[  1639][E][Wire.cpp:137] setPins(): bus already initialized. change pins only when not.
[  1639][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  1643][V][Touch_FT5x06.cpp:63] _check_init(): [FT5x06] CIPHER:0x64 / FIRMID:0x01 / VENDID:0x11
E (1377) ledc: ledc_get_duty(740): LEDC is not initialized

I have bought a mpu9255 module(no mpu9250 found). I connect it to esp32s3 with sda、scl、vcc and gnd.
However, when i check the data from uart, IMU initialization unsuccessful.
I check the sda scl  vcc and gnd via voltmeter，the volitage is ok with normal status.
Is reason that i cannot use mpu9250.h for the mpu9255 module? If it is right, i cannot find mpu9255.h, can you give me some links?

jgauchia, 2024-04-18:
Hello, this MPU is not supported. You need to keep in mind that this is a homebrew project, so I can't cover all possible combinations of elements. If you agree, what you're asking for requires a workaround, and if you're okay with it and don't think it's urgent, I've created a new branch to change the libraries so that it can include multiple IMU and compass modules. However, my time for dedication is limited.

zhjygit, 2024-04-19:
Thanks sir, i will have a try.

zhjygit, 2024-04-22:
ok. I changed the module to mpu9255 and "IMU initialization successful".

closed 2024-04-22 by zhjygit