jgauchia/IceNav-v3 #108: arrows direction and maps

zhjygit, 2024-04-22:
I have used the makerfabs to finish this project, everything is fine.
I used the mpu9250 module to receive compass data.
However, i found that the arrow on the maps cannot move, meanwhile, the maps move according to the compass data.
In common sense, like maps on the phone, the arrow direction changes with the owner walking direction, and the maps always remain unchangeable.
So, how can i hold the maps direction and change the arrow direction according to the compass?

jgauchia, 2024-04-22:
Hi, this feature isn't implemented yet. Actually navigation arrrow is centered at screen and map changes. I don't know when this feature will be available...

zhjygit, 2024-04-22:
ok,i will have a try.

zhjygit, 2024-04-29:
static void update_map(lv_event_t *event)
{
  CurrentMapTile = get_map_tile(getLon(), getLat(), zoom, 0, 0);

  if (strcmp(CurrentMapTile.file, OldMapTile.file) != 0 || CurrentMapTile.zoom != OldMapTile.zoom ||
      CurrentMapTile.tilex != OldMapTile.tilex || CurrentMapTile.tiley != OldMapTile.tiley)
  {
    is_map_draw = false;
    map_found = false;
    // map_spr.deleteSprite();
    // map_spr.createSprite(768, 768);
  }

  if (!is_map_draw)
  {
    OldMapTile.zoom = CurrentMapTile.zoom;
    OldMapTile.tilex = CurrentMapTile.tilex;
    OldMapTile.tiley = CurrentMapTile.tiley;
    OldMapTile.file = CurrentMapTile.file;

    log_v("TILE: %s", CurrentMapTile.file);
    log_v("ZOOM: %d", zoom);

    // Center Tile
    map_found = map_spr.drawPngFile(SD, CurrentMapTile.file, 256, 256);

    uint8_t centerX = 0;
    uint8_t centerY = 0;
    int8_t startX = centerX - 1;
    int8_t startY = centerY - 1;
    bool tileFound = false;

    if (map_found)
    {
      for (int y = startY; y <= startY + 2; y++)
      {
        for (int x = startX; x <= startX + 2; x++)
        {
          if (x == centerX && y == centerY)
          {
            // Skip Center Tile
            continue;
          }
          RoundMapTile = get_map_tile(getLon(), getLat(), zoom, x, y);
          tileFound = map_spr.drawPngFile(SD, RoundMapTile.file, (x - startX) * tileSize, (y - startY) * tileSize);
          if (!tileFound)
            map_spr.fillRect((x - startX) * tileSize, (y - startY) * tileSize, tileSize, tileSize, LVGL_BKG);
        }
      }
    }

    is_map_draw = true;
  }

  if (map_found)
  {
    NavArrow_position = coord_to_scr_pos(getLon(), getLat(), zoom);
    map_spr.setPivot(tileSize + NavArrow_position.posx, tileSize + NavArrow_position.posy);
    map_rot.pushSprite(0, 27);

#ifdef ENABLE_COMPASS
    heading = get_heading();
    //map_spr.pushRotated(&map_rot, 360 - heading, TFT_TRANSPARENT);
    map_spr.pushRotated(&map_rot, 0, TFT_TRANSPARENT);//保持地图不变
    map_rot.fillRectAlpha(TFT_WIDTH - 48, 0, 48, 48, 95, TFT_BLACK);
    map_rot.pushImageRotateZoom(TFT_WIDTH - 24, 24, 24, 24, 360 - heading, 1, 1, 48, 48, (uint16_t *)mini_compass, TFT_BLACK); 
#else
    map_spr.pushRotated(&map_rot, 0, TFT_TRANSPARENT);
#endif
    map_rot.setTextColor(TFT_WHITE, TFT_WHITE);

    map_rot.fillRectAlpha(0, 0, 50, 32, 95, TFT_BLACK);
    map_rot.pushImage(0, 4, 24, 24, (uint16_t *)zoom_ico, TFT_BLACK);
    map_rot.drawNumber(zoom, 26, 8, &fonts::FreeSansBold9pt7b);

    map_rot.fillRectAlpha(0, 342, 70, 32, 95, TFT_BLACK);
    map_rot.pushImage(0, 346, 24, 24, (uint16_t *)speed_ico, TFT_BLACK);
    map_rot.drawNumber((uint16_t)GPS.speed.kmph(), 26, 350, &fonts::FreeSansBold9pt7b);

    map_rot.fillRectAlpha(250, 342, 70, TFT_WIDTH - 245, 95, TFT_BLACK);
    map_rot.setTextSize(1);
    map_rot.drawFastHLine(255,360,60);
    map_rot.drawFastVLine(255,355,10);
    map_rot.drawFastVLine(315,355,10);
    map_rot.drawCenterString(map_scale[zoom], 285, 350);

    //sprArrow.pushRotated(&map_rot, 0, TFT_BLACK);
    sprArrow.pushRotated(&map_rot, heading, TFT_BLACK);//让导航箭头动起来，且与compass相反
  }

as above shown, i changed the code and the map tile remains silent, meanwhile, the arrow navigtion changes with the mpu9250 module.

jgauchia, 2024-05-07:
Disabling map rotation is already planned for the next version of the program.

zhjygit, 2024-05-08:
ok,thanks.

closed 2024-05-08 by zhjygit