jgauchia/IceNav-v3 #111: how can i get the real point north via the compass?

zhjygit, 2024-04-29:
There is a problem with the compass pointing north, and the compass in the upper right corner has nothing to do with the north, but only reflects the direction of movement.
The true pointing north is that no matter how much you move, the pointing north in the upper right corner always remains the same, that is, pointing north.

As you know, sir, i have finished that the navigation on the maps tile in the lcd center change direction according to the mpu9250 module.
so, i want to the compass in the upper right corner reflect the pointing north and remains silently to north,just like the phone.
how can complete it?

jgauchia, 2024-05-07:
The compass in map screen actually not show north, shows the position. 

Will keep this change you suggest in mind for future version; an option will be added to either keep the compass always pointing north or to function as it currently does.

jgauchia, 2024-05-14:
Added option in setting screen to enable/disable map compass rotation.
This feature will be available in the next stable release.

commit f4e0e109 (2024-05-14): FIx #111
closed 2024-05-14 by jgauchia