jgauchia/IceNav-v3 #116: Makefab missing libs

hpsaturn, 2024-05-14:
## Overview

In the current dev branch we had some minor issues

## Log output

```cpp
In file included from lib/gui/src/notifyBar.hpp:17,
                 from lib/lvgl/src/lvglSetup.hpp:20,
                 from src/main.cpp:44:
lib/bme/bme.hpp:12:10: fatal error: Adafruit_Sensor.h: No such file or directory
```

jgauchia, 2024-05-14:
Working on it!

jgauchia, 2024-05-14:
Updated dev branch, please, test it

hpsaturn, 2024-05-14:
Thanks. The build is working, and also it show the boot splash image, but then is black the screen, but without any panic:

```cpp
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   515][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   524][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=38 scl=39 freq=100000
[   524][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  4534][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: C_offset_x NOT_FOUND
[  4534][E][Preferences.cpp:503] getBytesLength(): nvs_get_blob len fail: C_offset_y NOT_FOUND
[  4539][V][Preferences.cpp:352] getUChar(): nvs_get_u8 fail: Map_rot NOT_FOUND
[  4547][V][Preferences.cpp:364] getShort(): nvs_get_i16 fail: GPS_speed NOT_FOUND
[  4553][V][Preferences.cpp:364] getShort(): nvs_get_i16 fail: GPS_rate NOT_FOUND
[  4561][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Coords_X NOT_FOUND
[  4568][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Coords_Y NOT_FOUND
[  4575][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Altitude_X NOT_FOUND
[  4582][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Altitude_Y NOT_FOUND
[  4589][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Speed_X NOT_FOUND
[  4596][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: Speed_Y NOT_FOUND
[  4603][V][Preferences.cpp:352] getUChar(): nvs_get_u8 fail: Map_vector NOT_FOUND
[  4610][V][settings.cpp:88] loadPreferences(): COMPASS OFFSET X  0.000000
[  4617][V][settings.cpp:89] loadPreferences(): COMPASS OFFSET Y  0.000000
[  4624][V][settings.cpp:90] loadPreferences(): MAP ROTATION 0
[  4629][V][settings.cpp:91] loadPreferences(): DEFAULT ZOOM LEVEL 16
[  4635][V][settings.cpp:92] loadPreferences(): SHOW MAP COMPASS 1
[  4641][V][settings.cpp:93] loadPreferences(): SHOW MAP SPEED 1
[  4647][V][settings.cpp:94] loadPreferences(): SHOW MAP SCALE 1
[  4653][V][settings.cpp:95] loadPreferences(): GPS SPEED 2
[  4658][V][settings.cpp:96] loadPreferences(): GPS UPDATE RATE 3
[  4664][V][settings.cpp:97] loadPreferences(): COMPASS POS X 12
[  4669][V][settings.cpp:98] loadPreferences(): COMPASS POS Y 158
[  4675][V][settings.cpp:99] loadPreferences(): COORDINATE POS X 15
[  4681][V][settings.cpp:100] loadPreferences(): COORDINATE POS Y 10
[  4687][V][settings.cpp:101] loadPreferences(): SPEED POS X 15
[  4693][V][settings.cpp:102] loadPreferences(): SPEED POS Y 100
[  4699][V][settings.cpp:103] loadPreferences(): ALTITUDE POS X 15
[  4705][V][settings.cpp:104] loadPreferences(): ALTITUDE POS Y 55
[  4710][V][settings.cpp:105] loadPreferences(): VECTOR MAP 0
[  4724][V][storage.cpp:33] initSD(): SD Card Mounted
[  4823][V][storage.cpp:50] initSPIFFS(): SPIFFS Mounted
[  5121][E][Wire.cpp:137] setPins(): bus already initialized. change pins only when not.
[  5122][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[  5125][V][Touch_FT5x06.cpp:63] _check_init(): [FT5x06] CIPHER:0x64 / FIRMID:0x03 / VENDID:0x11
E (4849) ledc: ledc_get_duty(740): LEDC is not initialized
[  5492][V][mainScr.cpp:89] editScreen(): 0
[  5492][V][mainScr.cpp:89] editScreen(): 0
[  5542][V][mainScr.cpp:89] editScreen(): 0
[  5543][V][mainScr.cpp:89] editScreen(): 0
[  5543][V][mainScr.cpp:89] editScreen(): 0
[  5544][V][mainScr.cpp:89] editScreen(): 0
[  5547][V][mainScr.cpp:89] editScreen(): 0
[  5551][V][mainScr.cpp:89] editScreen(): 0
[  5555][V][mainScr.cpp:89] editScreen(): 0
[  5559][V][mainScr.cpp:89] editScreen(): 0
[  5566][V][mainScr.cpp:89] editScreen(): 0
[  5567][V][mainScr.cpp:89] editScreen(): 0
[  5570][V][mainScr.cpp:89] editScreen(): 0
[  5577][V][mainScr.cpp:89] editScreen(): 0
[  5578][V][mainScr.cpp:89] editScreen(): 0
[ 14181][V][tasks.cpp:20] readGPS(): Task1 - Read GPS - running on core 1
[ 14182][V][tasks.cpp:21] readGPS(): Stack size: 18396
```

jgauchia, 2024-05-14:
Created a new branch in order to review LVGL settings for MakerFabs

jgauchia, 2024-05-15:
Problem with task & LVGL 9, Working on it

jgauchia, 2024-05-15:
Added a commit to the branch

hpsaturn, 2024-05-15:
Thanks! is working with the last version.

commit 2daae2d2 (2024-05-14): Fix #116 MakerFabs LCD
commit 1dcc1937 (2024-05-14): Fix #116 LVGL config
commit aca29847 (2024-05-14): Fix #116 MakerFabs LVGL
commit 7b4a173c (2024-05-14): Fix #116
closed 2024-05-15 by hpsaturn