jgauchia/IceNav-v3 #174: spi ili9488 or parallel ili9488

zhjygit, 2024-07-15:
In the platformio.ini:
[env:MAKERF_ESP32S3]:-D TFT_BL=45
void init_tft()
{
  tft.init();
  tft.setRotation(8);
  tft.initDMA();
  tft.startWrite();
  tft.fillScreen(TFT_BLACK);
  tft.endWrite();

  gpio_set_drive_capability(GPIO_NUM_33, GPIO_DRIVE_CAP_3);
  ledcAttachPin(TFT_BL, 0);

Now, my board is 16 bit parallel tft with driver FT6236, how can i define the drivers for it?
I can not find the driver setting for my 16 bit parrallel with dirver ft6236.
Waiting for you replaying.

jgauchia, 2024-07-15:
Parallel definition is here [](https://github.com/jgauchia/IceNav-v3/blob/master/lib%2Fpanel%2FILI9488_FT5x06_16B.hpp)

Do you need add buildflag in platformio, see readme

zhjygit, 2024-07-16:
> Parallel definition is here https://github.com/jgauchia/IceNav-v3/blob/master/lib%2Fpanel%2FILI9488_FT5x06_16B.hpp and here https://github.com/jgauchia/IceNav-v3/blob/master/include%2Fhal.hpp
> 
> Add buildflag in platformio, see readme

I can not understand.
At first, my board is  16 bit parallel tft with driver FT6236(not ft6x06);
In fact, without any change, now i can touch my board use the code of you, maybe it is slow touch feel.
If i want to add the support for the ili9488 with 16 bit parallel FT6236 board, i should create a FILI9488_FT6236_16B.hpp, and then where to include the FILI9488_FT6236_16B.hpp file?
Should i change the hal.hpp? what the relationship between hal.hpp and FILI9488_FT6236_16B.hpp?

At last, will i add a buildflag in .ino file? like -D ILI9488_FT6236_16B = 1? how will the project build it with -D? i cannot find it.

void init_tft()
{
tft.init();
tft.setRotation(8);
tft.initDMA();
tft.startWrite();
tft.fillScreen(TFT_BLACK);
tft.endWrite();

gpio_set_drive_capability(GPIO_NUM_33, GPIO_DRIVE_CAP_3);
ledcAttachPin(TFT_BL, 0);

what the relationship between above codes with my -D ILI9488_FT6236_16B = 1?

So boring, waiting for you replying.

jgauchia, 2024-07-16:
> > Parallel definition is here https://github.com/jgauchia/IceNav-v3/blob/master/lib%2Fpanel%2FILI9488_FT5x06_16B.hpp and here https://github.com/jgauchia/IceNav-v3/blob/master/include%2Fhal.hpp
> > 
> > Add buildflag in platformio, see readme
> 
> I can not understand.
> At first, my board is  16 bit parallel tft with driver FT6236(not ft5x06);

FT5x06 is a generic touch driver for i2c FT touch chip series. It's fully compatible

> In fact, without any change, now i can touch my board use the code of you, maybe it is slow touch feel.

How do you feel it's slow? I don't understand you, there are image in TFT? Yes or not?

> If i want to add the support for the ili9488 with 16 bit parallel FT6236 board, i should create a FILI9488_FT6236_16B.hpp, and then where to include the FILI9488_FT6236_16B.hpp file?

Why do you need yo create a new include file? There are one file for It, only do you need to check GPIO pin definitions un hal.hpp file

> Should i change the hal.hpp? what the relationship between hal.hpp and FILI9488_FT6236_16B.hpp?

In hal.hpp file are the gpio definition for the screen , pleade check It and made necessary changes If you need

> 
> At last, will i add a buildflag in .ino file? like -D ILI9488_FT6236_16B = 1? how will the project build it with -D? i cannot find it.

No, you need to ad buildflag at platformio.ini. un this file you set-up your environment (ESP model, GPS, etc..)

> 
> void init_tft()
> {
> tft.init();
> tft.setRotation(8);
> tft.initDMA();
> tft.startWrite();
> tft.fillScreen(TFT_BLACK);
> tft.endWrite();
> 
> gpio_set_drive_capability(GPIO_NUM_33, GPIO_DRIVE_CAP_3);
> ledcAttachPin(TFT_BL, 0);
> 
> what the relationship between above codes with my -D ILI9488_FT6236_16B = 1?

This is TFT init Code. This is independent from the panel or TFT choosed.

With the build-flag, you choose the Tft model and create the configuration for It.

> 
> So boring, waiting for you replying.
> 
>

closed 2024-08-09 by jgauchia