jgauchia/IceNav-v3 #192: Sating up MAKERFABS environment for ELECROW ESP32 Terminal

Shreyashsatone, 2024-08-28:
I am using board "ELECROW ESP32 Terminal". hear is the board link :-
https://robu.in/product/elecrow-esp32-terminal-with-esp32-3-5-inch-parallel-480x320-tft-capacitive-touch-rgb-displayrgb-by-chip-ili9488/?gad_source=1&gclid=Cj0KCQjwzva1BhD3ARIsADQuPnUBL5Wnh72beYLEnL0SMpd0_4TPQ-3lfNj_WxJGONgNmg5aiVz-tloaAmN6EALw_wcB

I was able to successfully deploy. But the Display of the board did not turne on and the code did not work no the ELECROW ESP32 Terminal.

jgauchia, 2024-09-04:
- [ ] Review Board definitions
- [ ] Add support to ELECROW ESP32 Terminal

jgauchia, 2024-09-05:
Branch https://github.com/jgauchia/IceNav-v3/tree/devel_support_elecrow_esp32 updated

https://youtube.com/shorts/3VFcUI6n7OQ

closed 2024-09-07 by jgauchia