jgauchia/IceNav-v3 #234: Add ESP-NOW connection between devices

jgauchia, 2024-10-24:
## Summary

Possibility of linking two or more devices via ESP-NOW, and sharing information.
For example, a master device that includes GPS and another device that displays the information on a screen, maps, etc...
The idea is to have a device in charge of collecting information related to GPS and/or sensors and another to display the information. It could be applied to smaller screens such as round screens, more portable devices, etc...

Brainstorming is welcome

zhjygit, 2025-02-24:
Is there a borad with GPS Integration? I bound makerfabs esp32se board with gps modules and battery, they are so fat.

jgauchia, 2025-02-24:
> Is there a borad with GPS Integration? I bound makerfabs esp32se board with gps modules and battery, they are so fat.

I don't know.

For instance , any board with serial GPS works