jgauchia/IceNav-v3 #260: Support for LILYGO T4 S3

il0vew33d, 2024-12-06:
please add LILYGO T4 S3. It has a 450x600 AMOLED touchscreen.

https://lilygo.cc/products/t4-s3

thanks a lot.

jgauchia, 2024-12-07:
Hi,
I'll try to add support for LilyGo T4 S3. I don't have this board, but I'll try it

jgauchia, 2024-12-16:
I'll try to give support for this board in the Next release, please be patient and stay tunned

jgauchia, 2024-12-17:
I've waiting for a T4 S3 Board, stay tunned for new updates to bring support for it

jgauchia, 2024-12-27:
Hi @il0vew33d , I’m waiting to receive a T4-S3 shortly to provide support as you requested. I just wanted to let you know that your issue is on my radar, and as soon as I receive the board, I'll get to work on it. I’d also like to mention that over the past few days, I’ve been looking into how it works, etc., and noticed that it’s not currently supported by the graphics library I use. That shouldn’t be a problem—on the contrary, it’s a challenge. Plus, I’m interested in providing this support since it has a lot of potential. I’ll keep you updated on my progress if you don’t mind, and I might ask you to validate some aspects of my firmware as I go along

il0vew33d, 2024-12-27:
hi jordi,
that's great news. what do you mean with graphics library? T4-S4 supports LVGL, I thought you also use LVGL?
https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series

of course  we can discuss some aspects of your firmware.
frank. :-)

jgauchia, 2024-12-27:
Yes, T4-S3 supports LVGL, but my project is based in LovyanGFX for all graphics, etc... This isn't a problem,I just ask you for a little patience; I’m interested in supporting as many boards as possible in my humble project hahaha.

jgauchia, 2024-12-27:
But don’t worry, with a bit of time, etc., I’ll add support. I’m interested in it

il0vew33d, 2024-12-27:
ah, ok. now I understand. please don't stress, we have time....

jgauchia, 2024-12-27:
hahahah, no problem... Quite the opposite, I’d like to make my firmware support different boards... of course, with the dedication I can give, as it’s a personal and/or hobby project

jgauchia, 2024-12-30:
hi @il0vew33d , I'm waiting to receive today a T4-S3.... I'm expecting to give support for this board... stay tunned...
Thank you

jgauchia, 2024-12-30:
[image]

Well, I've one... I'm goingo to give support, stay tunned.

jgauchia, 2025-01-02:
Hi @il0vew33d , I'm working on it...
There are some issues with this board and LovyanGfx graphics library than I use.

Open a discussion with them: https://github.com/lovyan03/LovyanGFX/discussions/663#discussioncomment-11711098

jgauchia, 2025-04-23:
Hi @il0vew33d , lovyangfx has a new branch to give support (limited) to T4-S3 .
I'm going to test it and try to give support to It in Icenav

jgauchia, 2025-04-25:
Some progress...

[image]

[image]

il0vew33d, 2025-10-26:
wouw, cool. that's good news. :-)

meanwhile i also have a lilygo t-deck plus and installed icenav. is there any description how it works?