jgauchia/IceNav-v3 #261: T-Deck restart after 2h of light sleep

hpsaturn, 2024-12-08:
## Overview

With the last implementations around the power consumption, was introduced an issue detected in T-Deck version that restart the device, maybe a PANIC or watchdog trigger, after of some hours en light sleep. This light sleep was launched using the button 0 or trackball button in the T-Deck

hpsaturn, 2024-12-08:
Thanks to this pull request #262, I found the exception and the cause of this issue:

The complete back trace and its threads are [here](https://termbin.com/jl3j), but I paste some lines here too. This exception was 3h after to to the light sleep in the makeup via GPIO 0 button.

```cpp
==================== CURRENT THREAD STACK =====================
#0  0x40378475 in panic_abort (details=0x3fc997ea "abort() was called at PC 0x4206f904 on core 0") at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
#1  0x4037ed18 in esp_system_abort (details=0x3fc997ea "abort() was called at PC 0x4206f904 on core 0") at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
#2  0x40385dc0 in abort () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
#3  0x4206f907 in task_wdt_isr (arg=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/task_wdt.c:176
#4  0x4037a5bc in _xt_lowint1 () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1118
#5  0x400559e0 in ?? ()
#6  0x40382281 in vPortClearInterruptMaskFromISR (prev_level=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
#7  vPortExitCritical (mux=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:332
#8  0x40379e11 in vPortExitCriticalSafe (mux=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:607
#9  timer_list_unlock (timer_type=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:346
#10 0x42092300 in timer_process_alarm (dispatch_method=ESP_TIMER_TASK) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:395
#11 timer_task (arg=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:422

======================== THREADS INFO =========================
  Id   Target Id          Frame 
* 1    process 1070542760 0x40378475 in panic_abort (details=0x3fc997ea "abort() was called at PC 0x4206f904 on core 0") at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  2    process 1070525548 0x4204e024 in gpsTask (pvParameters=<optimized out>) at lib/tasks/tasks.cpp:44
  3    process 1070549956 0x4217bf36 in esp_pm_impl_waiti () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_pm/pm_impl.c:855
  4    process 1070551356 0x4217bf36 in esp_pm_impl_waiti () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_pm/pm_impl.c:855
  5    process 1070514456 0x4200c7ca in loop () at src/main.cpp:202
  6    process 1070560056 cliTask (param=<optimized out>) at lib/tasks/tasks.cpp:69
  7    process 1070561128 0x4037fdea in vPortEnterCritical (mux=0x3fcf74ec) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
  8    process 1070535160 0x4037ff4c in vPortEnterCritical (mux=0x3fcf0dd0) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
  9    process 1070536860 0x4037ff4c in vPortEnterCritical (mux=0x3fcf1474) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
  10   process 1070524240 0x400559e0 in ?? ()

       TCB             NAME PRIO C/B  STACK USED/FREE
---------- ---------------- -------- ----------------
0x3fcf2fa8        esp_timer    22/22         640/3960
0x3fceec6c         GPS Task      1/1         656/7520
0x3fcf4bc4            IDLE0      0/0          608/412
0x3fcf513c            IDLE1      0/0          608/404
0x3fcec118         loopTask      1/1         576/7608
0x3fcf7338         cliTask       1/1        560/19424
0x3fcf7768          sys_evt    20/20         688/1860
0x3fcf11f8             ipc0    24/24          608/408
0x3fcf189c             ipc1    24/24          608/404
0x3fcee750         lvglDraw      3/3         688/7496

==================== THREAD 1 (TCB: 0x3fcf2fa8, name: 'esp_timer') =====================
#0  0x40378475 in panic_abort (details=0x3fc997ea "abort() was called at PC 0x4206f904 on core 0") at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
#1  0x4037ed18 in esp_system_abort (details=0x3fc997ea "abort() was called at PC 0x4206f904 on core 0") at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
#2  0x40385dc0 in abort () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
[... cut ...]
#9  timer_list_unlock (timer_type=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:346
#10 0x42092300 in timer_process_alarm (dispatch_method=ESP_TIMER_TASK) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:395
#11 timer_task (arg=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:422
```

jgauchia, 2024-12-08:
It's seems to be in esp_timer task, correct?

hpsaturn, 2024-12-09:
seems more that it is in the watchdog:
```cpp
esp-idf/components/esp_system/task_wdt.c:176
```

zhjygit, 2025-02-24:
Hi bro, you flash the code to t-deck?  Whether we can integrate the IceNav code to the t-deck firmware?Because the T-deck has keyboard, If we flash the to t-deck, we will not use the keyboard function or other many functions of t-deck firmware itself.

jgauchia, 2026-10-03:
@hpsaturn  still occurs, with new releases?

hpsaturn, 2026-10-06:
The issue persist. I think that now it is working better, but tonight (1:30am) I leave it, but now 8:00am, it resume but after that reboot..

[icenav_tdeck_coredump_20261006.zip](https://github.com/user-attachments/files/33090100/icenav_tdeck_coredump_20261006.zip)
[DIAG.log](https://github.com/user-attachments/files/33090045/DIAG.log)